OpenRoberta/openroberta-lab #469: Calliope: Play frequency block continues playing sound when simulation is stopped if duration is negative

Blippy2019, 2020-01-02:
**Describe the bug**
 - When play frequency block duration input is negative, the sound continues playing even after the simulation is stopped

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Calliope
2. Select play frequency block
3. Input negative value for duration
4. Start and stop simulation
5. See error

**Expected behavior**
 - No sound to be played or error message with block

**Screenshots**
 - 
[image]

**Device information**
 - Type: Desktop
 - OS: [e.g. windows], including version
 - Browser: Chrome

**Additional context**
 - Sound only stops when you refresh or close the tab. Switching robots or running another simulation does not stop the sound

Blippy2019, 2020-01-02:
This issue is also seen on EV3 and NXT

boonto, 2020-01-03:
Needs testing whether this is the case for the real robot as well.

boonto, 2020-01-03:
A very short sound is played on the real Calliope. Negative values should either not be able to be entered or the values should be clamped to the described range.

boonto, 2020-01-15:
Resolved by a GCI student in c16942a66225cb885169dd5c74deb3ad6a95cb38.

closed 2020-01-15 by boonto