OpenRoberta/openroberta-lab #470: Arduino Wifi Rev2: Incompatibility with some libraries (used for arduinos)

bjost2s, 2020-01-02:
**Describe the bug**
when e. g. an ultrasonic sensor or an infrared sensor is used in the configuration, the program cannot be compiled and a compiler error is displayed.

**TODO**
- Check carefully which sensors and actuators from the common arduino plugin can be used. 
- Try to find workarounds (other libraries) 
- If this is not possible we should decide how to support arduino wifi rev2 in the future, maybe a separate plugin is necessary

boonto, 2020-01-24:
The problematic sensors are RFID and infrared.
There is currently an open pull request to the infrared library [here](https://github.com/z3t0/Arduino-IRremote/pull/657) which should fix that one once merged and updated.
For the RFID sensors the problem seems to have been introduced in the 1.8.5 version of the megaavr core as documented [here](https://github.com/arduino/ArduinoCore-megaavr/issues/62).

boonto, 2020-01-24:
I also split the integration tests in #364 into their own "unowifirev2" folder, which should be removed and replaced with the "ardu" folder once the libraries work.

boonto, 2020-01-27:
As a quick workaround I added the "not supported" messages, so the user is notified that the two blocks do not work. No code is generated and compilation will succeed as a result.

boonto, 2020-08-28:
The IRremote library was updated to support Uno Wifi Rev2. I have updated our usages and removed the "not supported" messages.