OpenRoberta/openroberta-lab #471: Sensebox: Sensors used with List variable throws error when program is compiled

Caffetaria, 2020-01-02:
**Describe the bug**
 - When the List variable is used to store sensor values, and the program is compiled, an error is thrown.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new nepo program.
2. Create a new List Number variable.
3. Use sensors(which return numerical values) such as potentiometer, compass sensor and accelerometer as the elements of the list variable.
4. Press the compile button.
5. See error.

**Expected behavior**
Program should compile without errors.

**Screenshots**
[image]
[image]

**Device information**
 - Type : Desktop
 - OS: MacOS
 - Browser: Chrome

VinArt, 2020-01-03:
Seems to be already patched on develop

bjost2s, 2020-01-05:
@VinArt , please provide the related issue

RaghuvirShirodkar, 2022-10-13:
Seems to be solved. Cross compilation tested successful on `lab.open-roberta.org`.
Here is a [sample program](<https://lab.open-roberta.org/#loadProgram&&sensebox&&NEPOprog&&%3Cexport%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%3E%3Cprogram%3E%3Cblock_set%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%20robottype=%22sensebox%22%20xmlversion=%223.1%22%20description=%22%22%20tags=%22%22%3E%3Cinstance%20x=%22384%22%20y=%2250%22%3E%3Cblock%20type=%22robControls_start_ardu%22%20id=%22l9Lp%60gNd%5DGQJ_0PBfA?z%22%20intask=%22true%22%20deletable=%22false%22%3E%3Cmutation%20declare=%22true%22%3E%3C/mutation%3E%3Cfield%20name=%22DEBUG%22%3E%3C/field%3E%3Cstatement%20name=%22ST%22%3E%3Cblock%20type=%22robGlobalVariables_declare%22%20id=%2229yb@z%259%7CmaTv.%60,GM?7%22%20intask=%22true%22%20deletable=%22false%22%20movable=%22false%22%3E%3Cmutation%20next=%22false%22%20declaration_type=%22Array_Number%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3Eitem%3C/field%3E%3Cfield%20name=%22TYPE%22%3EArray_Number%3C/field%3E%3Cvalue%20name=%22VALUE%22%3E%3Cblock%20type=%22robLists_create_with%22%20id=%22e6rLmt%5EkP~j.%7DaALRR;V%22%20intask=%22true%22%3E%3Cmutation%20items=%2210%22%20list_type=%22Number%22%3E%3C/mutation%3E%3Cfield%20name=%22LIST_TYPE%22%3ENumber%3C/field%3E%3Cvalue%20name=%22ADD0%22%3E%3Cblock%20type=%22robSensors_light_getSample%22%20id=%22pK:!S%7C=To(e(Fov)uBJP%22%20intask=%22true%22%3E%3Cmutation%20mode=%22LIGHT%22%3E%3C/mutation%3E%3Cfield%20name=%22MODE%22%3ELIGHT%3C/field%3E%3Cfield%20name=%22SENSORPORT%22%3EL%3C/field%3E%3Cfield%20name=%22SLOT%22%3E%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ADD1%22%3E%3Cblock%20type=%22robSensors_sound_getSample%22%20id=%22Ha0x5V%7CYtrJJz:5qq%7CZj%22%20intask=%22true%22%3E%3Cmutation%20mode=%22SOUND%22%3E%3C/mutation%3E%3Cfield%20name=%22MODE%22%3ESOUND%3C/field%3E%3Cfield%20name=%22SENSORPORT%22%3ES2%3C/field%3E%3Cfield%20name=%22SLOT%22%3E%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ADD2%22%3E%3Cblock%20type=%22robSensors_ultrasonic_getSample%22%20id=%22lEHDuax%7D/#OW-?Cdisuv%22%20intask=%22true%22%3E%3Cmutation%20mode=%22DISTANCE%22%3E%3C/mutation%3E%3Cfield%20name=%22MODE%22%3EDISTANCE%3C/field%3E%3Cfield%20name=%22SENSORPORT%22%3EU%3C/field%3E%3Cfield%20name=%22SLOT%22%3E%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ADD3%22%3E%3Cblock%20type=%22robSensors_humidity_getSample%22%20id=%22lW_~LIcR!(p7yzEWplie%22%20intask=%22true%22%3E%3Cmutation%20mode=%22HUMIDITY%22%3E%3C/mutation%3E%3Cfield%20name=%22MODE%22%3EHUMIDITY%3C/field%3E%3Cfield%20name=%22SENSORPORT%22%3EH%3C/field%3E%3Cfield%20name=%22SLOT%22%3E%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ADD4%22%3E%3Cblock%20type=%22robSensors_temperature_getSample%22%20id=%22%5Ds=N0zDTnA81n02%7D~_th%22%20intask=%22true%22%3E%3Cmutation%20mode=%22TEMPERATURE%22%3E%3C/mutation%3E%3Cfield%20name=%22MODE%22%3ETEMPERATURE%3C/field%3E%3Cfield%20name=%22SENSORPORT%22%3ET%3C/field%3E%3Cfield%20name=%22SLOT%22%3E%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ADD5%22%3E%3Cblock%20type=%22robSensors_lightveml_getSample%22%20id=%22!AZ~%5Dc3Q)/ML%7C.4G*6PW%22%20intask=%22true%22%3E%3Cmutation%20mode=%22LIGHT%22%3E%3C/mutation%3E%3Cfield%20name=%22MODE%22%3ELIGHT%3C/field%3E%3Cfield%20name=%22SENSORPORT%22%3EV%3C/field%3E%3Cfield%20name=%22SLOT%22%3E%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ADD6%22%3E%3Cblock%20type=%22robSensors_accelerometer_getSample%22%20id=%2223rfL!9/R2U8Dat1)fLE%22%20intask=%22true%22%3E%3Cmutation%20mode=%22X%22%3E%3C/mutation%3E%3Cfield%20name=%22MODE%22%3EX%3C/field%3E%3Cfield%20name=%22SENSORPORT%22%3EB%3C/field%3E%3Cfield%20name=%22SLOT%22%3E%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ADD7%22%3E%3Cblock%20type=%22robSensors_particle_getSample%22%20id=%22!=,h5xZp%7C=--B%5ESQ.fbO%22%20intask=%22true%22%3E%3Cmutation%20mode=%22PM25%22%3E%3C/mutation%3E%3Cfield%20name=%22MODE%22%3EPM25%3C/field%3E%3Cfield%20name=%22SENSORPORT%22%3ES%3C/field%3E%3Cfield%20name=%22SLOT%22%3E%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ADD8%22%3E%3Cblock%20type=%22robSensors_gps_getSample%22%20id=%22jhMcqVZ;n%25r7,uu)97f1%22%20intask=%22true%22%3E%3Cmutation%20mode=%22LATITUDE%22%3E%3C/mutation%3E%3Cfield%20name=%22MODE%22%3ELATITUDE%3C/field%3E%3Cfield%20name=%22SENSORPORT%22%3EG2%3C/field%3E%3Cfield%20name=%22SLOT%22%3E%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ADD9%22%3E%3Cblock%20type=%22robSensors_environmental_getSample%22%20id=%22%60j@#r)!F?Jm%5Bc-KE%5DH=i%22%20intask=%22true%22%3E%3Cmutation%20mode=%22TEMPERATURE%22%3E%3C/mutation%3E%3Cfield%20name=%22MODE%22%3ETEMPERATURE%3C/field%3E%3Cfield%20name=%22SENSORPORT%22%3EE%3C/field%3E%3Cfield%20name=%22SLOT%22%3E%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/statement%3E%3C/block%3E%3Cblock%20type=%22robControls_loopForever_ardu%22%20id=%22%60mIJB4z:(3D;T%25:%5D?%5EIz%22%20intask=%22true%22%20deletable=%22false%22%20movable=%22false%22%3E%3Cstatement%20name=%22DO%22%3E%3Cblock%20type=%22robControls_forEach%22%20id=%22C#R7i_!,K=#Ae%25*v?@eW%22%20intask=%22true%22%3E%3Cmutation%20list_type=%22Number%22%3E%3C/mutation%3E%3Cfield%20name=%22TYPE%22%3ENumber%3C/field%3E%3Cfield%20name=%22VAR%22%3Eitem2%3C/field%3E%3Cvalue%20name=%22LIST%22%3E%3Cblock%20type=%22variables_get%22%20id=%22YNDT7Ae%7BAgMurHirXGrA%22%20intask=%22true%22%3E%3Cmutation%20datatype=%22Array_Number%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3Eitem%3C/field%3E%3C/block%3 [... cut ...] 22robConf_gps%22%20id=%22K_XC-;gG?QdV5%60=BZe;D%22%20intask=%22true%22%3E%3Cfield%20name=%22NAME%22%3EG2%3C/field%3E%3Cfield%20name=%22I2C%22%3EI2C%3C/field%3E%3C/block%3E%3C/instance%3E%3Cinstance%20x=%2218%22%20y=%22810%22%3E%3Cblock%20type=%22robConf_environmental%22%20id=%22_l%5EceCPsa)yKN6:%5EPNo%5D%22%20intask=%22true%22%3E%3Cfield%20name=%22NAME%22%3EE%3C/field%3E%3Cfield%20name=%22I2C%22%3EI2C%3C/field%3E%3C/block%3E%3C/instance%3E%3C/block_set%3E%3C/config%3E%3C/export%3E>) for testing.

closed 2022-10-24 by bjost2s