OpenRoberta/openroberta-lab #473: Calliope: Set brightness with parameter 0 not working

KevinL10, 2020-01-02:
**Describe the bug**
 - The description of the block "setBrightness" says that "0 means all leds are turned off, 9 is the brightest value." However, when the brightness is set to 0, it has the same brightness as if the brightness was set to 9. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to a calliope program
2. Add a setBrightness block with brightness 0
3. Display a text
4. See error

**Expected behavior**
 - Since "0 means all leds are turned off" a setBrightness block with parameter 0 should have all leds turned off. 

**Screenshots**
 - This program will produce the error 
[image]

**Device information**
 - Type: Desktop
 - OS: Windows 10
 - Browser: Chrome

**Additional context**
 - Note that there is a similar block, which sets the brightness for a specific LED. That block will 
 work correctly(not display anything) when brightness is set to 0.
- This program will work correctly 
[image]

boonto, 2020-01-03:
Needs testing whether this is the case for the real robot as well.

boonto, 2020-01-03:
0 brightness works correctly on the real Calliope.

boonto, 2020-01-06:
Resolved by GCI student in 4b71015.

closed 2020-01-06 by boonto