OpenRoberta/openroberta-lab #474: Microbit: Add setBrightness and getBrightness blocks

KevinL10, 2020-01-02:
**Is your feature request related to a problem? Please describe.**
 - Calliope mini and Calliope mini 2016 have setBrightness and getBrightness blocks which set the brightness for the entire screen as opposed to a single LED.
- 
[image]

**Describe the solution you'd like**
 - Add these blocks to micro::bit -> expert blocks -> Action -> Display.

RohanCheri, 2020-01-04:
When checking the MciroBit (Python) API, I was actually unable to find the set and get brightness functions. 
(I have attached a link to the API here: https://microbit-micropython.readthedocs.io/en/latest/display.html) 

Instead, I was thinking that we could implement the microbit.display.on(), off(), is_on(), and read_light_level(), which is already a function made in Calliope.

nishanth1232, 2020-01-07:
The Microbit Python API (MicroPython - https://github.com/OpenRoberta/micropython ) unfortunately does not have the setbrightness and get brightness functions as mentioned by @RohanCheri 

We could also define the function itself as such:
```
function set_brightness(value):
  for i = 0 in 4
  for j = 0 in 4
set_pixel(i, j, value)
```
and,
```
function get_brightness(value):
  for i = 0 in 4
  for j = 0 in 4
  get_pixel(i, j)
```
The only problem with the code is that the set_pixel function turns on the LED while setting the brightness as well but the set_brightness function is supposed to only set the brightness of the LED and turn it on when something is displayed on it.