OpenRoberta/openroberta-lab #476: EV3: stopping simulator does not stop sound playing

Blippy2019, 2020-01-03:
**Describe the bug**
 - Stopping the simulator has no effect on any sound playing. The sound will continue until it is finished even if the simulation is stopped. When the 'Say' block is used with a long sentence, the robot will finish the sentence even when the simulation is stopped halfway through.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to EV3 simulation
2. Use a play frequency block and set duration to 10 seconds
3. Start the simulation and stop it immediately
4. See error - sound keeps on playing until 10 seconds pass

**Expected behavior**
 - Sound should stop immediately when simulation is stopped

**Screenshots**
 - 
[image]
[image]

**Device information**
 - Type: Desktop
 - OS: Windows 10
 - Browser: Chrome

boonto, 2020-01-03:
Needs testing whether this is the case for the real robot as well.

boonto, 2020-01-03:
0 brightness works correctly on the real Calliope.

boonto, 2020-01-03:
EV3dev: The program will stop, but the speech output will still continue in the background.

ensonic, 2020-01-03:
How did you stop the program on EV3dev - with middle+down buttons?

boonto, 2020-01-03:
Yes, that should be the same behaviour as the stop button in the simulation. On monday I will try it on lejos and c4ev3 as well and check if it is the same there. If it is the same on every platform the simulation is just accurate :)