OpenRoberta/openroberta-lab #480: EV3: Negative Volumes Being Stored

RohanCheri, 2020-01-03:
**Describe the bug**
When setting the volume to a negative number and using get volume, a negative number is displayed, which shouldn't be possible

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Action -> Sounds
2. Set Volume to a negative number
3. Display the Volume
4. Observe the negative value

**Expected behavior**
 The volume value being clipped to a range between 0 and 100 (The program would display 0) OR not allowing negative values entirely in the set volume block.

**Screenshots**
[image]

**Device information**
 - Type Desktop
 - OS: Windows 10
 - Browser Chrome 70.0.3945.88

**Additional context**
The problem occurs in all EV3 Devices.

boonto, 2020-01-03:
Needs testing whether this is the case for the real robot as well.

boonto, 2020-01-07:
Does not happen on dev or lejos.

rbudde, 2020-03-19:
works in the simulation

closed 2020-03-19 by rbudde