OpenRoberta/openroberta-lab #481: EV3: Simulation Reset Brick Light Not Working Properly

RohanCheri, 2020-01-03:
**Describe the bug**
When using the "Reset Brick Light," the description states that it sets the default bricklight, "green and blinking," but when the block is used, it just resets the bricklight, similar to the "brick light off" function.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Action -> Lights and add block "colour [green] [on]" (Any color works)
2. Add wait after block
3. Add  "reset brick light" after wait
4. Add another wait at the end
5. Complie and run program
6. Observe Behavior
7. (Optional) Replace "reset brick light' with "brick light off" and observe the exact same behavior.

**Expected behavior**
 The bricklight being reset to its default state (green and blinking), as stated in the description of the "reset brick light" block.

**Screenshots** (Initial Bricklight setting has been changed to orange for testing purposes)
Behavior of "Reset Brick" block: 
[image]

Behavior of "Brick Light Off" block:
[image]

**Device information**
 - Type Desktop
 - OS: Windows 10
 - Browser Chrome 79.0.3945.88

**Additional context**
 -Occurs with all EV3 Devices

boonto, 2020-01-03:
Needs testing whether this is the case for the real robot as well.

boonto, 2020-01-06:
Seems to be the case on ev3dev as well: #242.
We need to discuss whether the tooltip or the behaviour is wrong.