OpenRoberta/openroberta-lab #484: Show robot type of imported program

boonto, 2020-01-03:
**Is your feature request related to a problem? Please describe.**
- When importing a program into the Lab, which does not fit the currently selected robot type, you get an error message stating incompatibility.
- The user does not know which robot type the imported program has, or to which the user would have to switch.

**Describe the solution you'd like**
 - The robot type of the imported program is shown in the message.

boonto, 2020-01-03:
Already partially resolved by a GCI student in #460.

boonto, 2020-01-06:
As the exported XML programs only contain the robot group and not a specific robot type, something specialised has to be displayed here.
There are three options:
1. Show a default robot (EV3Lejos V1 for ev3, Arduino Uno for arduino). This may confuse inexperienced users (C4EV3 users may get "This is for EV3Lejos" and think this program will not work with C4EV3).
2. Add a "real" name for robot groups to the properties. This would lead to duplicated code in the properties.
3. Show all members of group with suitable message ("The program may be for one of these robots: ..., ..., ...").

I prefer option 3.

boonto, 2020-01-07:
Option 3 was implemented, translators will be notified through @philippmaurer that messages changed.

bjost2s, 2020-01-07:
There are still issues with the messages:
English:
[image]

German:
[image]
please make sure it looks like this at the end, Guillemet are part of our CI!
... »EV3dev, EV3 c4ev3, EV3 leJOS 0.9.0, EV3 leJOS 0.9.1« ...

VinArt, 2020-01-09:
English looks fine now

closed 2020-01-09 by VinArt