OpenRoberta/openroberta-lab #493: Microbit: true != null causes unexpected behaviour

Caffetaria, 2020-01-04:
**Describe the bug**
 - When an 'if' block is used with the condition true != null, the do block does not execute.
This is not the expected behaviour in usual programming languages.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a nepo program.
2. Use the if block.
3. Use the condition "true != null".
4. Do statement does not execute in the simulation.

**Expected behavior**
 - The 'do' statement of the 'if' block should execute as "true != null" evaluates to true.

**Screenshots**
- 
[image]

[image]

**Device information**
 - Type: Desktop
 - OS: MacOS
 - Browser: Chrome

**Additional context**
 - Seems to happen in other robot simulations as well.

bjost2s, 2020-02-12:
should be fixed together with #306

RaghuvirShirodkar, 2022-10-04:
Seems to be solved with the new simulation.

[image]

bjost2s, 2022-10-04:
The problem is not solved, because "if true == null" produces same result.

PR #1377 (2022-10-20): Issue #493: fix condition `true == null` causes incorrect behaviour in simulation
# Description
This PR implements the following changes:

Fix null const leads to simulation errors:

- update visitNullConst in AbstractStackMachineVisitor.
- add case for C.NULL_CONST in interpreter.
- correct tests to accommodate changes.

Fixes #493 

## Type of change

Please delete options that are not relevant.

- [x] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?
Tested with programs in simulation containing the `if true == null` block.
Integration tests run successfully as [... cut]
closed 2022-10-24 by bjost2s