OpenRoberta/openroberta-lab #502: EV3: Set variable block with missing arguments throws a server error

Caffetaria, 2020-01-05:
**Describe the bug**
 - When a variable is created and the 'set' block for that variable is used without an input, the user receives a server error.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new nepo program.
2. Create a new 'Number' variable.
3. Use the 'set' variable block without an input.
4. Open the simulation
5. See error

**Expected behavior**
 - The user should get an alert that says "There are errors in your program. Please check the messages" and the block should show a warning that says "An input value is missing!".

**Screenshots**
 - 
[image]

**Device information**
 - Type: Desktop
 - OS: MacOS
 - Browser: Chrome

**Additional context**
 - The error is also seen in other Calliope, NXT and Micro:bit.

boonto, 2020-01-05:
Will probably be solved by #306.

boonto, 2020-01-13:
Server error for now is resolved by a GCI student.

closed 2020-01-13 by boonto