OpenRoberta/openroberta-lab #508: Microbit wait until temperature sensor returns False

Blippy2019, 2020-01-07:
**Describe the bug**
 - The temperature sensor from the dropdown in microbit's wait until block returns False. The sensor view shows that the temperature is 25, but the sensor returns the False value instead.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to expert blocks
2. Find wait until <sensor value> block
3. Set sensor to temperature sensor
4. Run simulation
5. See error

**Expected behavior**
 - The temperature sensor should return 25

**Screenshots**
 - Simulation with code below should display heart, however the heart never shows.
[image]
 - When the sensor block is used with a show text block, the temperature value is displayed as "false"
[image]

**Device information**
 - Type: Desktop
 - OS: Windows 10
 - Browser: Chrome

**Additional context**
 - Error was tested on https://test.open-roberta.org/# as well as https://lab.open-roberta.org/#
 - Problem also occurs on Calliope simulation

boonto, 2020-01-07:
Does this exclusively happen for the "wait until" block, or the standard block as well?

Blippy2019, 2020-01-07:
This only happens to the wait until block, the normal sensor returns the right value 25.