OpenRoberta/openroberta-lab #517: Simulation sayText does not work on iOS

boonto, 2020-01-08:
**Describe the bug**
The simulation sayText block does not work on iOS. Blocks used after the block are also never processed.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a say block
2. Add a drive block after the say block
3. Open the simulation
4. Run the program

**Expected behavior**
 - The robots says the message and starts to drive afterwards.

**Device information**
 - Type: iPad Pro (9,7 Zoll) 
 - OS: IOS 13.1.3
 - Browser: Safari, Chrome

bjost2s, 2020-01-09:
Works fine on IOS Iphone, but only separated from tones. When a play tone and a say block is in the same program the play tone block is only once executed. The next start of the same program in the simulation suppresses the play note (not the say ;-))

boonto, 2020-01-10:
It works with tones now (most of the time, sometimes it inexplicably won't work for some reason, even after reloading the page).

bjost2s, 2020-01-12:
Tested on IOS Iphone with Safari and Chrome

closed 2020-01-12 by bjost2s