OpenRoberta/openroberta-lab #525: Software Requirements Specification document missing

kartecianos, 2020-01-09:
Hi, people !

I noticed that there isn't any SRS document and I thought it would be nice to have SRS. It could be used as documentation by end users and developers too.

Should I start making it?

boonto, 2020-01-15:
What exactly do you mean by that? Do you mean hardware requirements? The software requirements needed to compile all robots are listed in the README.md.

kartecianos, 2020-01-15:
You can find some guidelines and a description of what is a SRS document here: https://www.perforce.com/blog/alm/how-write-software-requirements-specification-srs-document

In general, it is a 30 pages long doc in which there is all the information of what exactly the software does.

boonto, 2020-01-20:
Oh, I didn't expect requirements engineering. It would be nice to have, but I think we would have to do it ourselves, as I assume it is quite hard to specify these requirements from an outside view.

kartecianos, 2020-01-20:
It is hard to specify requirements from an outside view. That's why requirements engineer have to communicate with the members of the project to take the information needed. I am not a professional nor do I have much experience on requirements engineering. I just use open roberta a lot and I am interested in requirements engineering.

We could try it and make a SRS document (your part would be giving me information) and you could decide if you want to keep it or not. However you know better what's good for the project.

boonto, 2020-01-28:
That's true. I think this should come from our side at some point, thanks for the suggestion!