OpenRoberta/openroberta-lab #528: NXT: Only the first light-sensor with shining LED

corvus-albus, 2020-01-10:
**Describe the bug**
Given a NXT-brick with two connected light-sensors, only the sensor which is first mentioned in the program will have shining LED, the second one has LED switched off. Both sensors are questioned to show light (and not ambient light). 
I changed cables, ports and tested to different NXT-bricks. The problem is always the same.
 
**To Reproduce**
Steps to reproduce the behavior:
1. Take a NXT-brick and connect two NXT-light-sensors. 
2. Write a little program, for example
[image]
3. Upload the program to the brick
4. Start program
5. Only the LED of the first mentioned lightsensor will be switch on, the other one will be dark.
6. Change the order of the sensor-blocks, for example:
[image]
7. Again ony the first LED will shine.

**Expected behavior**
Both sensor should have the LED switched on.

**Device information**
 - Type: Desktop
 - OS: windows 7
 - Browser: firefox 72.0.1 (64-Bit)

boonto, 2020-01-15:
I don't see any immediate issue from the code generation or the implemenation of `_readLightSensor(..)`. This would require some more thorough investigation.

corvus-albus, 2020-01-16:
I got it working by adding an extra line in the sourcecodeeditor (thank you for this new feature!). 

Openroberta translate the blocks to:
#define WHEELDIAMETER 5.6
#define TRACKWIDTH 12.0
#define MAXLINES 8
#include "NEPODefs.h" // contains NEPO declarations for the NXC NXT API resources

task main() {
    SetSensor(S1, SENSOR_LIGHT);
    SetSensor(S4, SENSOR_LIGHT);
    while ( ! ButtonPressed(BTNCENTER, false) ) {
        NumOut(0, (MAXLINES - 1) * MAXLINES, _readLightSensor(S1, 1));
        NumOut(0, (MAXLINES - 2) * MAXLINES, _readLightSensor(S4, 1));
    }
}
That has the effect described above.

I added one line:
    ...
    SetSensor(S4, SENSOR_LIGHT);
    **SetSensorType(IN_4, SENSOR_TYPE_LIGHT_ACTIVE);**
    ...
Afterwards it was working. Both LEDs are set to "ON".

Cheers corvus-albus

boonto, 2020-01-16:
Very nice that the source code editor helped to find a possible solution.

The `SetSensorType` should actually happen inside of the `_readLightSensor()` [method](https://github.com/OpenRoberta/ora-cc-rsc/blob/master/RobotNXT/NEPODefs.h#L810), however the mode is cached in the `_LIGHT_SENSOR_MODE` variable and the mode setup is not done for the second one.