OpenRoberta/openroberta-lab #537: Source Code Editor: scrollbars are covered by the text area

bjost2s, 2020-01-14:
**Describe the bug**
 see screenshot

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'open source code editor'
2. Make your window smaller, so that the vertical and the horizontal scrollbars appear
3. scroll until you observe what is in the screenshot

**Expected behavior**
 Scrollbars should be outside the text area

**Screenshots**
[image]

Ubuntu Chrome

harshitmalik03, 2020-11-11:
Can you show me the file to work on this.

KishkinJ10, 2021-03-23:
i am a beginner cna i work on this issue ? can you  tell me from where should i start

CarefoDragneel, 2021-12-11:
I would like to work on this issue. Actually, I am an open-source enthusiast but I am new in this field. So, can you please guide me on how to understand the codebase and exactly how to make my contribution here?

bjost2s, 2021-12-29:
@CarefoDragneel , it seems to be an issue of bad / wrong css, Are you familiar with debugging in a browser? That might help a lot

spandit230496, 2023-01-19:
i am interested can you assign this issue to me

bjost2s, 2023-03-24:
works as expected

closed 2023-03-24 by bjost2s