OpenRoberta/openroberta-lab #540: Edison: Play frequency block only works with constant values over 244

philippmaurer, 2020-01-14:
**Describe the bug**
If you use a dynamic calculation or a constant value of 244 or lower as a frequency in Hz for the "play frequency" block, the program will not compile.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to ORLab and select Edison
2. Put the "play frequency" block into your program and change the Hz value either to a value below 245, or to a calculation that uses a variable or sensor.
3. Run program
4. See error

**Expected behavior**
The ORLab should compile the programs.

**Additional context**
If you use a constant calculation whichs result is below 245, the program will still compile.

boonto, 2020-01-23:
Once this bug is fixed the block should be reactivated in the integration tests added with #364.

ranajay99, 2020-06-23:
Hi I'm a newbie. Can you give me some lead how to work on this issue?

bjost2s, 2020-06-24:
Please ask in our gitter channel  https://gitter.im/openroberta-lab/community, it is  for us much easier to answer to general questions in the chat.

rbudde, 2020-06-24:
the generated code contains a sequence 8000000/FREQ_GIVEN_BY_USER. If FREQ_GIVEN_BY_USER is a constant, the expression is evaluated by the COMPILER. If it is variable, it is evaluated on the robot.

Edison only allows values -32,767 to +32,767. Thus: if the COMPILER evaluates the expression to a result within these limits, everything is ok, otherwise the compiler detects, that a number has to be stored, that is too large and complains.

BTW: an Edison program with one variable set to 60000 fails for the same reason. The tone is no problem at all :-)

rbudde, 2023-04-11:
works

closed 2023-04-11 by rbudde