OpenRoberta/openroberta-lab #541: Multiple robot simulation: Dropdown menu for selecting robots in sensor view does not display the options

Caffetaria, 2020-01-14:
**Describe the bug**
 - When using the multiple robot simulation, the dropdown menu in the sensor's data view does not display the different robot options. Instead a blank dropdown menu is displayed. The colour of the robot is shown only when an option from the dropdown menu is selected.

**To Reproduce**
Steps to reproduce the behavior:
1. Create 2 or more nepo programs and use them in multiple robot simulation.
2. Go to the sensor's data view.
3. Click on the dropdown menu to change the robot.
4. See error

**Expected behavior**
 - The dropdown menu should display the different robot options.

**Screenshots**
 - 
[image]

**Device information**
 - Type: Desktop
 - OS: MacOS
 - Browser: Chrome

KevinL10, 2020-01-18:
This issue is reproducable only under certain conditions:
**Windows**

-Chrome: cannot be produced

-Firefox: cannot be produced

-Internet Explorer: cannot be produced

**MacOs**
-Chrome: reproducable

-Firefox: cannot be produced

-Safari: cannot be produced

**Ubuntu**

-Firefox: reproducable 

The html is generated here:
[image]

bjost2s, 2023-06-05:
fixed with #907

closed 2023-06-05 by bjost2s