OpenRoberta/openroberta-lab #546: Multiple robot simulation: Switching from simulation to source code view and back to simulation causes the multiple simulation to end

Caffetaria, 2020-01-15:
**Is your feature request related to a problem? Please describe.**
 - When using the multiple simulation, if the user switches from the multiple simulation to the source code view and then back to the simulation view, the multiple robot simulation ends and only one robot remains.

**Describe the solution you'd like**
 - The user should be given an option to end the multiple simulation when switching to the source code view.

**Describe alternatives you've considered**
 - The user can be notified that the multiple robot simulation has ended when shifting to the source code view.

**Additional context**

[image]

KevinL10, 2020-01-20:
This issue also applies to the other views, such as "program documentation view" and the "help view".

sachinsshetty, 2020-04-02:
Same issue raised closed as not a bug,  https://github.com/OpenRoberta/openroberta-lab/issues/787 

Multiple simulation window is single focus view to run different programs, When switched to source code view, current program is brought into focus and simulation switches to the current program robot execution.

boonto, 2020-04-06:
This one is a feature request, not a bug report. Here it is proposed that the closing of the multiple sim is handled more gracefully, as not to confuse users why the multiple sim is gone afterwards.