OpenRoberta/openroberta-lab #547: Arduino compilation without arduino-builder

VinArt, 2020-01-15:
Arduino compilation takes long time due to recompilaton of the sketch. The idea is alike to #170 and #34 to use pre-compiled libraries and cores.

All needed libraries can be deduced from the blockly blocks and precompiled with arduino-builder compiling one über-sketch with all includes.

A GccCompiler worker can be used to invoke gcc for fast (~0.1s) compilation time

boonto, 2020-03-26:
I have started to upgrade everything to use `arduino-cli` instead of `arduino-builder`, this way we can use dependency management for all of the libraries we don't modify and reduce a lot of the bulk of committed `.h`  and `.cpp` which didn't have any version information and were very tough to maintain. I'll also add Festo and Bot'n Roll.

- [ ] Restructure to use `arduino-cli`
- [ ] Add Festo (ESP32)
- [ ] Add Bot'n Roll

boonto, 2020-04-20:
Transfer onto a Uno Wifi Rev. 2 does not work right now. `avrdude` exits with `verification error; content mismatch` when flashing bootloader. Looking into it.

boonto, 2020-04-20:
Fixed by adding a linker flag and linking `variant.c.o`.

bjost2s, 2020-04-22:
All related robots have tested, compilation and execution, works very well!

closed 2020-04-22 by bjost2s