OpenRoberta/openroberta-lab #552: Persistent pin state for Calliope/micro:bit simulation

boonto, 2020-01-16:
**Is your feature request related to a problem? Please describe.**
 - Currently the state of the pins (analog/digital in) in the Calliope/micro:bit simulation is reset every simulation start. This is sometimes unexpected for users and may not accurately reflect usage of the pins in real situations.

**Describe the solution you'd like**
- PIN states are persistent for the session.

PamoCurtis, 2023-04-28:
Since the release of the new simulation, this issue in not present anymore. But quite the opposite issue is now present: The pins for __analog/digital out__ from the calliope/micro:bit(v2) aren't resetted after every simulation start.

bjost2s, 2023-05-08:
tested

closed 2023-05-08 by bjost2s