OpenRoberta/openroberta-lab #555: Multiple Robot Simulation: Over 10 bots.

KevinL10, 2020-01-17:
**Describe the bug**
 - When the user chooses over 10 robots to be part of a multiple robot simulation, only one robot will be shown, which is desired. However, the sensor data dropdown menu will still have 9 colours to choose from.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a multiple robot simulation with over 10 robots
2. Go to sensor data
3. Open the dropdown
4. See error

**Expected behavior**
 - Either alert the user that there cannot be more than 9 robots in a multi simulation, or make the dropdown show only one colour.

**Screenshots**
 -
[image]

**Device information**
 - Type: Desktop
 - OS: Windows 10
 - Browser: Chrome

boonto, 2020-01-17:
May be a duplicate of #557.

boonto, 2020-03-06:
In order to solve this we should limit the selectable amount of robots to 9.

sachinsshetty, 2020-04-02:
Issue not reproducible in version 3.8.5.
After selecting 12 program in multiple robot simulation, the dropdown in sensor view shows all 12 color for robots.
Tested on Ubuntu 18.04, firefox 74.0.

bjost2s, 2023-06-05:
fixed with #907

closed 2023-06-05 by bjost2s