OpenRoberta/openroberta-lab #556: Multiple Robot Simulation: FPS showing incorrect value

KevinL10, 2020-01-17:
**Describe the bug**
 - If you create a multiple robot simulation with over 3 robots, the FPS in the sensor data section will show an alternation between 500 and 1000 fps. If you then create another multiple robot simulation, the fps will alternate between Na, 500, and 1000 fps.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a multiple robot simulation program with over 2 robots
2. Open sensor data
3. See error
4. Create another multiple robot simulation with over 2 robots
5. Open sensor data
6. See new error

**Expected behavior**
 -  The actual fps should be shown.

**Screenshots**
 - 
[image]

**Device information**
 - Type: Desktop
 - OS: Windows 10
 - Browser: Chrome

kuruma99, 2020-03-20:
Hi, I would like to work on this as my first bug.

kuruma99, 2020-03-25:
Hi @bjost2s , sorry could not work on this due to some issues. Will start working on this right away. Is there any more info regarding this that I should know. Thanks.

roffinator3000, 2022-03-11:
Please assign me

closed 2022-04-04 by bjost2s

rbudde, 2022-05-04:
works as expected

PR #1236 (2022-04-04): Issue #556: invisible simulations are canceled  
A running rendering of a simulation is now always canceled before starting a new one.
New frames are only requested after the pending frame is fully rendered.

Fixes #556 

## Type of change
- Bug fix (non-breaking change which fixes an issue)

## How Has This Been Tested?
- maven integration Tests
- visual tests in the lab