OpenRoberta/openroberta-lab #557: Multiple Robot Simulation

KevinL10, 2020-01-17:
**Describe the bug**
 - If you create a multiple robot simulation and afterwards you open a new program, the dropdown in sensor data will still contain the multiple colours of robots.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a multiple robot simulation with over 1 robot.
2. Close that multiple simulation by opening source code or going to "edit" -> "my programs" and selecting any program.
3. Open the sensor data
4. See error

**Expected behavior**
 - There should only be one colour for that one robot.

**Screenshots**
 - 
[image]

**Device information**
 - Type: Desktop
 - OS: Windows 10
 - Browser: Chrome

boonto, 2020-01-17:
There should actually be no dropdown at all.

ykabusalah, 2020-03-17:
Can I try doing this?

boonto, 2020-03-17:
Sure!

ykabusalah, 2020-03-22:
> Sure!

I'll try to get it done soon! Can I get assigned this?

closed 2021-08-12 by bjost2s