OpenRoberta/openroberta-lab #560: Multiple robot simulation: The commands on the EV3 display are not distributed among the robots.

RichardBotez, 2020-01-17:
**Describe the bug**
If you create a multiple robot simulation and command the first robot to go forwards and the second robot to go backwards by pressing enter on the EV3 display, only one robot will receive the command.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a program for each robot, one should go forwards and one backwards by pressing enter on the EV3 display;
2. Run multiple robot simulation;
3. Simulate the program and press enter on the EV3 screen;
4. See error.

**Expected behavior**
 - Robots should run in opposite directions.

**Screenshots**
 - Link to the video: https://youtu.be/pkEBr8v-PeA
[image]

**Device information**
 - Type Desktop;
 - OS: Windows 10;
 - Browser: Chrome

bjost2s, 2020-01-18:
please click on the second robot‘s screen as well. View it by activating it with a double click on the orange robot first.

boonto, 2020-03-06:
Should have worked the way @bjost2s described at the time the issue was created, but was broken when fixing #565.

closed 2020-03-18 by rbudde