OpenRoberta/openroberta-lab #565: Multiple robot views in EV3 when new backgrounds are uploaded

Caffetaria, 2020-01-18:
**Describe the bug**
 - When two or more backgrounds are uploaded for EV3,  multiple robot views are displayed.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new EV3 program
2. Upload two or more new backgrounds
3. Open the simulation window.
4. Open the robot view.
5. See error.

**Expected behavior**
 - Only one robot view should show up.

**Screenshots**
 - 
[image]

**Device information**
 - Type: Desktop
 - OS: MacOS
 - Browser: Chrome

**Additional context**
 - The issue appears in EV3 but is solved for NXT.

boonto, 2020-01-22:
Resolved by a GCI student.

boonto, 2020-03-06:
The current implementation leads to #560.

boonto, 2020-03-17:
Does not happen anymore.

closed 2020-03-17 by boonto