OpenRoberta/openroberta-lab #566: 'Keep Me Logged In' Checkbox

nishanth1232, 2020-01-18:
**Is your feature request related to a problem? Please describe.**
 - It gets annoying when the user gets logged out of the lab by refreshing the page.

**Describe the solution you'd like**
 - A 'Keep Me Logged In' checkbox in the login menu.

**Describe alternatives you've considered**
 - A popup that asks permission for automatically logging in the next time when the lab is opened.

**Additional context**
 - This could also fix #559

Caffetaria, 2020-01-18:
Good idea because in this case the user is implicitly giving permission for cookies to be allowed. Or to be explicit the message might look like this:
**"Keep Me Logged In (Allow Cookies)"**

nishanth1232, 2020-01-18:
@Caffetaria 
That works too.

boonto, 2020-03-06:
Thanks for the suggestion!
Although, it would be easier for individual users, many of our users are based in schools. Here, leaving such a tick mark checked and forgetting to log out (which can happen easily with lessons ending) might allow unwanted access to your account and programs.

closed 2020-03-06 by boonto