OpenRoberta/openroberta-lab #567: 'Square' Mathematical Function

nishanth1232, 2020-01-18:
**Is your feature request related to a problem? Please describe.**
 - For all the robots, the 'math' blocks contain the square root function but not the square function.

**Describe the solution you'd like**
 - A block with the square function.

**Describe alternatives you've considered**
 - A square function added to the drop-down list in the square root 
 block.

**Additional context**
 - The list contains the square-root function but not the 'square' function.
[image]

rbudde, 2020-03-19:
seems to work, tested for Calliope, ev3 all variants

closed 2020-03-19 by rbudde

bjost2s, 2020-03-19:
also tested on micro:bit