OpenRoberta/openroberta-lab #579: Addition of Robot Information in Welcome Screen

nishanth1232, 2020-01-18:
**Is your feature request related to a problem? Please describe.**
 - Not all robots in the welcome screen have an "_i_" icon on them which redirects the user to the manufacturer's website (or) a page consisting of the robot's information.

**Describe the solution you'd like**
 - Addition of the icons which redirects the user to the manufacturer's website for robots that do not have one.

**Describe alternatives you've considered**
 - Icon that redirects the user to a page consisting of the robot's information (such as Wikipedia).

bjost2s, 2020-03-06:
This is only the case for calliope, which has sub plugins but they are visible in the welcome screen (exeptionally).
We should add the calliope "i" on the top level

boonto, 2020-04-24:
Fixed in previous release.

closed 2020-04-24 by boonto