OpenRoberta/openroberta-lab #582: SenseBox Information in German

nishanth1232, 2020-01-18:
**Describe the bug**
 - When the information for SenseBox is opened from the welcome note in the lab, it redirects to the German version of the manufacturer's website.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the welcome note.
2. Click on the "_i_" in SenseBox.
3. See website in German.

**Expected behavior**
 - It should redirect to the English version of the manufacturer's website.

**GIF**
 - 
[image]

**Device information**
 - Type: Laptop
 - OS: Windows 10
 - Browser: Google Chrome

boonto, 2020-01-30:
As we have most of our users in Germany the best solution would actually be a link based on the selected language.

nishanth1232, 2020-01-30:
@boonto 
But, the user cannot select the language in the welcome screen since it is the first screen to open. So, a redirection to the English website should be fine I think. If there is a German version of the website, the user can switch to it as the manufacturer's website will probably have an option to do so.

boonto, 2020-01-30:
The initial language is the one set by the browser or operating system.

nishanth1232, 2020-01-30:
Made the changes in the 'Expected Behaviour' Section.

boonto, 2020-04-24:
Fixed in previous release.

closed 2020-04-24 by boonto