OpenRoberta/openroberta-lab #585: Popup on uploading background does not show correct behaviour

Caffetaria, 2020-01-18:
**Describe the bug**
 - When uploading a background, a popup(i.e. The Open Roberta Lab can automatically load your simulation background on your next visits....) is displayed. 
However clicking on either 'Okay' or 'Cancel' causes the same behaviour of bringing up system files so that you can upload your background.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new EV3 on NXT program
2. Open the simulation window.
3. Click on the button to upload a new simulation background.
4. Click on Cancel when the popup  message is displayed
5. See error.

**Expected behavior**
 - Clicking the 'Cancel' button of the popup message should not pull up the system files to upload the background. 

**Screenshots**
 -
[image]

**Device information**
 - Type: Desktop
 - OS: MacOS
 - Browser: Chrome

**Additional context**
 - The problem can be seen on the develop branch.

boonto, 2020-01-19:
The intended behaviour is described in #38. Both "OK" and "Cancel" will open the file selection, as the popup just lets the user decide whether to use local storage or not.

Maybe we should change the wording of "OK" and "Cancel" to clarify that.

boonto, 2020-01-20:
As this is closely related to #38 further updates will be there.

closed 2020-01-20 by boonto