OpenRoberta/openroberta-lab #596: MicroBit: Missing Sensor

nishanth1232, 2020-01-19:
**Describe the bug**
 - According to the manufacturer(https://microbit.org/guide/features/), the MicroBit robot has a light sensor in it but the option for the sensor is not available in the lab.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch to MicroBit.
2. Click on Sensor's category.
3. See missing light sensor.

**Expected behavior**
 - Availability of light sensor in the sensor's list.

**Screenshots**
 - 
[image]

**Device information**
 - Type: Laptop
 - OS: Windows 10
 - Browser: Google Chrome

rbudde, 2020-03-17:
works. But the light sensor returns always 100% light. It should be possibile to set the value in the sensor data panel in the same way as for other sensors.

bjost2s, 2020-03-17:
Please try to hover with the mouse over the led matrix, the value is then 50 (like shadowed) , if you touch down with the mouse on the led matrix, light becomes brighter and the value is 150. This was the first sensor implemented in the simulation (no extra panel available at that time ).

closed 2020-03-19 by bjost2s