OpenRoberta/openroberta-lab #601: Lab Shows Project Information in German

nishanth1232, 2020-01-19:
**Describe the bug**
 - In the lab, if the help button is clicked and the 'about the Open Roberta Project' option is clicked, it opens a website in German.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the lab.
2. Click on the 'Help' button.
3. Click on 'about the Open Roberta Project' option.
4. See page in German.

**Expected behavior**
 - Page should be in English.

**GIF**
 - 
[image]

**Device information**
 - Type: Laptop
 - OS: Windows 10
 - Browser: Google Chrome

rohit-bindal, 2020-01-22:
I'll work on this!

boonto, 2020-02-17:
Thanks for the find, will be tracked in another issue tracker.

closed 2020-02-17 by boonto