OpenRoberta/openroberta-lab #603: Right-Clicking In the Lab Opens Two Menu's

nishanth1232, 2020-01-19:
**Describe the bug**
 - Sometimes, while right-clicking in the lab, it opens up the browser menu as well as the lab menu with the browser menu appearing ontop of the lab menu.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the lab.
2. Keep Right-Clicking.
3. See the bug.

**Expected behavior**
 - Only the lab menu should open.
(or)
Both menus should not be on-top of each-other.

**GIF**
 - 
[image]

**Device information**
 - Type: Laptop
 - OS: Windows 10
 - Browser: Google Chrome

RaghuvirShirodkar, 2022-10-04:
Not reproducible. Maybe solved...?
Tested on Firefox and Chrome.

bjost2s, 2022-10-04:
Seems to be a browser problem, that we cannot solve. Occurs not always!