OpenRoberta/openroberta-lab #604: Undo Does Not Work After Deleting a Block

nishanth1232, 2020-01-19:
**Describe the bug**
 - In the lab, pressing Undo (or) Ctrl+Z after deleting a block does not make the block appear again.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the lab.
2. Make a program consisting of 2-3 blocks.
3. Delete one block and press Undo.
4. See the bug.

**Expected behavior**
 - The block should come back into the program.

**GIF**
 - 
[image]

**Device information**
 - Type: Laptop
 - OS: Windows 10
 - Browser: Google Chrome

boonto, 2020-01-19:
May be a blockly problem.

MarkusBordihn, 2022-09-08:
This seems still a reproducible issue in 2022.

Given the fact it's working fine inside the Blockly Playground and other Blockly based implementation, it looks more like an integration issue with the Open Roberta Lab and Blockly and not like a Blockly issue itself.