OpenRoberta/openroberta-lab #605: Edison: Labelled Robot Configuration

nishanth1232, 2020-01-19:
**Describe the bug**
 - The robot configuration of Edison has the top of the robot labelled, but the picture of the bottom of the robot containing the power switch and line tracking sensor is not shown.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch to Edison.
2. Click on Robot Configuration.
3. See that only the top of the robot is shown and labelled.

**Expected behavior**
 - Both the top and bottom of the robot should be labelled.
It should look like this:

[image]

[image]

**Screenshots**
 - 
[image]

**Device information**
 - Type: Laptop
 - OS: Windows 10
 - Browser: Google Chrome

boonto, 2020-01-19:
Just the power switch is missing, which is not a huge problem. I think the real problem is that it is only available in german.

nishanth1232, 2020-01-19:
I guess this issue can be closed if it is not a problem. It is related to #571

closed 2020-01-19 by nishanth1232