OpenRoberta/openroberta-lab #612: EV3 and NXT: Arrow keys cause unexpected behaviour

Caffetaria, 2020-01-20:
**Describe the bug**
 - The left and right arrow keys make the robot turn left and right respectively. This is expected.
The expected behaviour for the up and down arrow keys would be to make the robot move front and back respectively.
However, the up and down arrow keys make the robot turn left and right respectively( similar to the left and right keys).

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new EV3 or NXT program
2. Open the simulation
3. Test the behaviour of the arrow keys
4. See error

**Expected behavior**
 - The up and down arrow keys should make the robot move front and back respectively.

**Screenshots**
 - Left Arrow key
[image]

- Right Arrow Key
[image]

- Up Arrow Key
[image]

- Down Arrow Key
[image]

**Device information**
 - Type: Desktop
 - OS: MacOS
 - Browser: Chrome

KevinL10, 2020-01-23:
I have been able to prevent the up and down keys from turning, but cannot find a way to make the robot move forward. Here's the current progress, someone else can build on this.
[image]

Caffetaria, 2020-01-23:
Cool @KevinL10! 
I think I got it working. 
Can I make a pull request?

Caffetaria, 2020-01-23:
@KevinL10 I believe that it may require some translation to the global coordinate system ... but I could be wrong.

Caffetaria, 2020-01-23:
@KevinL10 
Here's where I am at the moment. 
Doing something like this works in moving the robot back on hitting the down arrow:

```
case "ArrowDown":
                robots[robotIndex].pose.x -= 1;
                robots[robotIndex].pose.theta -= Math.PI / 180;
```
However, this does not work if the robot is not oriented horizontally. The change in theta is required in order to correct for a 'drift' that occurs in the orientation of the robot. 

Will update if I learn more.

boonto, 2020-01-24:
Resolved in d7afa3a58686cbcfb368eb296f51dd448b114d04.

closed 2020-01-24 by boonto