OpenRoberta/openroberta-lab #614: Adding bitwise operators

KevinL10, 2020-01-20:
**Is your feature request related to a problem? Please describe.**
 - It would be great to add bitwise operators ( &, ^, |, ~) in the expert blocks under math.

**Describe the solution you'd like**
 - Add a block, similar to the math blocks (+, -, x, / ), except for the purposes shown above.

KevinL10, 2020-01-30:
@boonto  Would you like (&, ^, |, ~) in the same math block as +, -, x, /, or in a separate block specifically for these operators? And which block would you like (~) to be in, since it only takes in one parameter?

boonto, 2020-01-30:
~ would fit somewhat in the "square root" block. The others should definitely be in another block as the +,- block is in the beginner toolbox.