OpenRoberta/openroberta-lab #618: Documentation View: Font Color Does Not Work

nishanth1232, 2020-01-21:
**Describe the bug**
 - In the documentation view, if the font color is changed, the font does not change.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the lab.
2. Click on Documentation View.
3. Switch to different font colors and check the font.
4. See the bug.

**Expected behavior**
 - Font color should change.

**GIF**
 - 
[image]

**Device information**
 - Type: Laptop
 - OS: Windows 10
 - Browser: Google Chrome

boonto, 2020-01-21:
A text needs to selected beforehand, then the selected changes will be applied to the selected text and the text written afterwards.

rohit-bindal, 2020-01-21:
I've fixed #593 issue opened by you. Have a look!
And I'll work on this issue also :)

boonto, 2020-02-17:
This seems to be a bug of the WYSIWYG library we currently use. A fix in that or using another library entirely would be necessary. We might reopen in the future.

closed 2020-02-17 by boonto