OpenRoberta/openroberta-lab #624: Pressing 'Help' on Some Image Blocks Does Not Redirect To Anything

nishanth1232, 2020-01-21:
**Describe the bug**
 - In the images category, some blocks such as the 'shift image' block and the 'invert' image block when right-clicked, do not redirect to anything in the 'help' view.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the lab.
2. Click on images.
3. Spawn the 'shift image' block, right-click it, and press 'Help'
4. See incorrect redirection.

**Expected behavior**
 - It should redirect to the correct title in the 'Help' View.

**GIF**
 - 
[image]

**Device information**
 - Type: Laptop
 - OS: Windows 10
 - Browser: Google Chrome

nishanth1232, 2020-01-21:
#570 
#597 
#616

bjost2s, 2020-02-17:
The help view 's content is generated from our wiki. We track that outside of github. Thank you very much.

closed 2020-02-17 by bjost2s