OpenRoberta/openroberta-lab #626: Play Note Block Having Incorrect Notes

RohanCheri, 2020-01-21:
**Describe the bug**
 - In the play note block, the note "H" is put in the place where the note "B" should be.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Actions and drag out the play note block
2. Click on the note and observe the error in the piano popup

**Expected behavior**
The note should say "b" instead of "h"

**Screenshots**
Current Display:
[image]

What the Note Should Be: 
[image]

**Device information**
 - Type Desktop
 - OS: Windows 79
 - Browser Chrome

**Additional context**
 - Occurs for all instances of where the note should be (Any octave)

boonto, 2020-01-21:
H is the German name for the note "B". This should be translated according to the language.

RohanCheri, 2020-01-21:
Oh, that's cool I did not know that.