OpenRoberta/openroberta-lab #627: List Function In The Wrong Area

RohanCheri, 2020-01-21:
**Describe the bug**
 - For all robots that support more complex math and list operations, the block "sum of list" and other options is in the Math section instead of the list section as it is a list function.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to math
2. Observe that the list block is located there

**Expected behavior**
 - For the list block to be in the "lists" category

**Screenshots**
[image]

**Device information**
 - Type Desktop
 - OS: Windows 10
 - Browser Chrome 79

**Additional context**
 - Appears across multiple robots

boonto, 2020-01-21:
As these are "math" operations on "lists" we chose to put them in the Math toolbox instead of the Lists toolbox. We need to discuss whether we keep it this way or change it.

RohanCheri, 2020-01-21:
I suggested this since all of the other blocks in the math section are only able to take a "Number" variable/block while this block takes in a list of numbers as an input... I see why it would be in the math section though since it only applies to lists of numbers.

bjost2s, 2020-02-17:
This is derived from blockly origin. We want to keep it as it is.

closed 2020-02-17 by bjost2s