OpenRoberta/openroberta-lab #630: Stop option does not work

Caffetaria, 2020-01-21:
**Describe the bug**
 - The stop option does not work.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new EV3 program
2. Open the simulation
3. Start the simulation
4. Open the background navigation dropdown.
5. Click on the stop option
6. See error

**Expected behavior**
 - The simulation should stop.

**Screenshots**
 - 
[image]

**Device information**
 - Type: Desktop
 - OS: MacOS
 - Browser: Chrome

**Additional context**
 - Additional information for the dropdown options should be given through a tooltip.

KevinL10, 2020-01-24:
@boonto Is the purpose of the stop button to stop the program? If so, can we export the function of the "start the program in simulation" and use it for the stop button?

boonto, 2020-01-27:
I believe it should have the same behaviour as clicking the stop button in the bottom left of the simulation, which appears while the program is running.

boonto, 2020-02-13:
Sorry again, the whole menu should be removed, see #745.

closed 2020-02-13 by boonto