OpenRoberta/openroberta-lab #631: 'Back' option does not work

Caffetaria, 2020-01-21:
**Describe the bug**
 - The 'Back' option in background navigation dropdown does not work.

Steps to reproduce the behavior:

1. Create a new EV3 program
2. Open the simulation
3. Open the background navigation dropdown.
4. Change the backgrounds using the background navigation view.
5. Click on the 'Back' option
6. See error

**Expected behavior**
 - The background should switch to the previous background.

**Screenshots**
 -
[image]

**Device information**
 - Type : Desktop
 - OS: MacOS
 - Browser: Chrome

**Additional context**
More information should be provided on the function of these options in the form of a tooltip or help option.

KevinL10, 2020-01-27:
Could I create a stack variable and add to the stack everytime there is a new background, and pop from the stack everytime the back button is pressed? And if the stack has size 0 then the back button should be 'disabled' and not be able to be clicked? @boonto

boonto, 2020-01-30:
We decided it would be best to just remove the back button, as it does not fit with the current usage of the simulation.

boonto, 2020-02-13:
Sorry for the back and forth but we decided to remove the whole menu, see #745.

closed 2020-02-13 by boonto