OpenRoberta/openroberta-lab #632: In certain cases, the scene dropdown icon appears when it is not supposed to

Caffetaria, 2020-01-21:
**Describe the bug**
 - The scene dropdown icon usually appears only when the simulation is opened.
However in certain scenarios, it appears even when the simulation is closed which appears inconsistent to the user.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new EV3  or NXT program
2. Open the simulation.
3. Switch to the source code view
4. Close the view
5.See error

**Expected behavior**
 - When the simulation is closed, the scene dropdown icon should not appear.

**Screenshots**
 - Usual behavior
[image]

- Unexpected behavior
[image]

**Device information**
 - Type: Desktop
 - OS: MacOS
 - Browser: Chrome

boonto, 2020-01-21:
Seems to be connected to #443.

sardug10, 2021-02-03:
Hey, I am new to open-source contributions. I want to work on this issue. Can you guide me?