OpenRoberta/openroberta-lab #636: Arduino: Sensor "No Port" Configuration Not Showing Error

RohanCheri, 2020-01-21:
**Describe the bug**
 - When running the Arduino robot, the user is able to open the code view/compile the code without any errors even when the sensor port is not recognized / it hasn't been added to the configuration file.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to an Arduino robot
2. Add a "Show on serial monitor" block and attach any sensor block to it
3. Observe that there is no port for the sensor
4. Open the code view/compile the code and observe that no error occurs

**Expected behavior**
 - For the program to throw an error and indicate that the configuration is missing.

**Screenshots**
[image]
[image]

**Device information**
 - Type [Desktop/tablet/smartphone]
 - OS: [e.g. windows], including version
 - Browser [e.g. chrome, firefox], including version

**Additional context**
 - If a motor block is attempted to be compiled without a port, the program properly throws an error.

boonto, 2020-04-29:
This seems to have been mostly fixed beforehand, the only missing blocks were "get value sensor" and "write value actuator".

bjost2s, 2020-05-07:
works fine now, also for the named blocks above!

closed 2020-05-07 by bjost2s