OpenRoberta/openroberta-lab #646: EV3 Simulation: Adding Clear Screen Button

RohanCheri, 2020-01-21:
**Is your feature request related to a problem? Please describe.**
 - When using the EV3 simulation, specifically on the ones where you can draw on the background, there is no way to clear the screen of all the marks other than going to a different background and coming back to it.

**Describe the solution you'd like**
 - A button that clears all drawings on the screen

**Describe alternatives you've considered**
 - It could also possibly be made into a block that is only usable in the simulation (Meaning that it doesn't show up in code view)

boonto, 2020-02-13:
We really don't want to add any more buttons to the simulation UI. It can probably be combined with the "reset position" button to provide a complete "reset" functionality.

bjost2s, 2023-04-28:
solved

closed 2023-04-28 by bjost2s