OpenRoberta/openroberta-lab #662: EV3: The Calibrate Block is Not Considered As a Blocking Block

nishanth1232, 2020-01-22:
**Describe the bug**
 - For the EV3, the 'calibrate HT compass sensor' block is not considered as a blocking block even though it takes 40 seconds to complete.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch to EV3.
2. Spawn the 'calibrate HT compass sensor' block.
3. See that it is not considered as a blocking block.

**Expected behavior**
 - A warning showing that it is a blocking block.

**GIF**
 - 
[image]

**Device information**
 - Type: Laptop
 - OS: Windows 10
 - Browser: Google Chrome

iamyashm, 2020-03-10:
I would like to work on this. Is this a purely frontend bug?

boonto, 2020-03-10:
Sure, block related changes need to be done over in our [blockly](https://github.com/OpenRoberta/blockly) repo and then built as described and copied over to the static resources of the server.

For first quick tests you can use your browser developer tools to edit the files and immediately see the changes.

iamyashm, 2020-03-10:
I've figured out code changes. Where can I find the build instructions for the blockly repo?

boonto, 2020-03-10:
It was previously documented in the upstream repo or the link, but as that changed a lot we should add it to our repo.
You can just run `python build.py` in the blockly directory while having the closure library on the same level as the blockly directory.
[image]

iamyashm, 2020-03-10:
I tried that but build.py in running into an exception.

[image]

boonto, 2020-03-10:
You should be able to try again, sometimes the build server is unreliable.

iamyashm, 2020-03-10:
Should I directly make the required change in staticResources/blockly instead of building and copying, since doing so seems to be working locally?

iamyashm, 2020-03-10:
Yes, it worked. Thank you.

boonto, 2020-03-11:
Fix merged in https://github.com/OpenRoberta/blockly/commit/a7b5caffd53f9879e88d2f10f4dc630b6ebf592a. Is resolved once blocky is rebuilt.

closed 2020-03-17 by bjost2s