OpenRoberta/openroberta-lab #667: BOB3 trigonometric functions & timer compilation fails

boonto, 2020-01-22:
**Describe the bug**
 - Programs containing trigonometric functions or the timer block can't compile.

**To Reproduce**
Steps to reproduce the behavior:
1. Add sin block
2. Compile the program
3. See error
 ---

1. Add get timer block
2. Compile the program
3. See error

**Expected behavior**
 - The programs compile.

**Additional context**
 - Probably a remainder of #170.

RohanCheri, 2020-01-22:
By any chance, could you attach a video of this error... I have not been able to reproduce it.

boonto, 2020-01-23:
[image]
Here is a screenshot of the compiler output when running the program. The hardcoded "PI" is missing after the changes of #170.

RohanCheri, 2020-01-23:
[image]
Oh okay thanks, from what I can tell, it seems like the issue comes from the generated code when the user uses a trigonometric function to be just "PI" instead of "M_PI." This would explain the error as the error states that the variable is not defined in this scope.

RohanCheri, 2020-01-24:
Additionally, I have been unable to recreate the error with the timer function so is it possible that you could also attach a screenshot or video of that error?
[image]
(This compiles fine for me)

boonto, 2020-01-24:
That exact program fails for me, you can try in the Lab with the NOCONNCT token.

RohanCheri, 2020-01-24:
Just a question, where would this NOCONNCT token be located?

boonto, 2020-01-24:
If you want to connect to a robot via the "connect ..." option in the robot dropdown you can enter the `NOCONNCT` token to simulate a connection to the robot.
[image]
This way you can compile without having the robot connected.

With the new source code editor you could also just click on "Build" which is a little bit simpler.

RohanCheri, 2020-01-25:
@boonto So I have been able to recreate the error with the timer and I have a proposed cause of the problem and solution. When looking at the error report, it seems that the function "millis()," which is an Arduino function, has never been defined. This may be due to the fact that the Arduino libraries have never been implemented in the Cross-Compiler, which would lead to the program to throw this issue.
One solution to this problem is to just define millis() in a different way and have it refer to the amount of time measured from "Epoch time."
This could be done by implementing the following code in the "bob3.h" file:
```
#include <chrono>

uint64_t millis() {
  using namespace std::chrono;
  return duration_cast<milliseconds>(system_clock::now().time_since_epoch()).count();
}
```

From there, rather than referring to the time function provided by Arduino, the robot can rely on time using the milliseconds since "EPOCH' which doesn't require much of a change other than defining the function above in the bob3.h file.

Another solution is to actually implement the Arduino libraries into the compiled code, which could also help with other functions but may take much longer than just implementing the code shown above.

boonto, 2020-01-27:
We had used the Arduino libraries before #170 that's where we got `millis()` from.
Now they are not included anymore and the functionality is missing.

I believe `std::chrono` won't work as it is not available for BOB3. We would have to find another solution.

RohanCheri, 2020-01-27:
After researching the topic, it seems as if there is no built in function to access time like there is in Java with system.currentTimeMillis() in C++.
This would mean that we need to include some library involving time one way or the other in the cross compiler (it doesn’t need to be downloaded onto the robot because there the millis() function would be declared). You said that chrono would not work so some other time libraries would be crime.

RohanCheri, 2020-01-27:
ctime*

boonto, 2020-01-29:
We'll have to check what would actually work on BOB3.

RohanCheri, 2020-01-29:
I meant that this would be like a part of the simulation itself because in the actual robot the milis() function would work as expected since the robot has access to the arduino library.

It could just be an in-built part of the compiler that isn’t actually downloaded onto the robot.

boonto, 2020-01-29:
The robot will just run whatever is in the compiled binary for it. The Arduino library is not included anymore, which means we don't have access to `millis()` anymore, I am unsure whether `ctime` is available in the standard lib the BOB3 currently uses.

RohanCheri, 2020-01-29:
Is there any way that we could just take the part of the arduino library where the `millis()` function is defined?

boonto, 2020-01-30:
I am not really sure. @VinArt what do you think?

VinArt, 2020-01-31:
provide a tiny millis implementation and see if it would work? May ask Nils himself what he thinks.

boonto, 2020-02-17:
Trigonometric functions should be handled when merging changes of #278. Thanks to RohanCheri for the help.

rbudde, 2020-03-19:
the program compiles, the timing is ok.

But there is a subtle difference to the old version. Even if the timer is not used at all (see the program attached), the LED is flickering, even if it is switched off. Not very bright, but slightly. I think, it is acceptable, but we should have a look onto that in the future.

Move to 'next version'
[image]

rbudde, 2020-03-19:
PLEASE DO NOT CLOSE FOR 3.8.4

bjost2s, 2020-04-19:
@rbudde , timer and trigonometic functions are working, still a problem?

rbudde, 2020-04-20:
working. I extended the math_functions integration tests by all trig functions. Working. I added a timer test. Working.

closed 2020-04-20 by rbudde