OpenRoberta/openroberta-lab #668: Collapsing 'Start' Block Adds 'FALSE' Text

nishanth1232, 2020-01-22:
**Describe the bug**
 - For certain robots such as WeDo, if the 'start' block is collapsed, additional 'FALSE' text appears.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch to WeDo.
2. Collapse the 'start' block.
3. See 'FALSE' text.

**Expected behavior**
 - No additional text should appear.

**GIF**
 - 
[image]

**Device information**
 - Type: Laptop
 - OS: Windows 10
 - Browser: Google Chrome

iamyashm, 2020-03-11:
Hi. Where can I find the code for the collapse event handler?

boonto, 2020-03-11:
Should be somewhere in blockly.